lumapu/ahoy #1845: HMS-1000-2T not reconise

PaTiTan, 2025-06-21:
## Hardware
  * [ ] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32 WROOM
Retailer URL: https://amzn.eu/d/ioXKoqS

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.155
Github Hash: 5feb293

### Build & Flash Method:
* [X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

My AhoyDTU has been running for 2 years with: 
2x HM-800
1x HMS-2000

And I would like to install my new HMS-1000-2T
Does the HMS-1000-2T with this serial use a frame format not yet supported?

Gubi2023, 2025-06-21:
are you shure you dont have a HMS-1000**W**-2T WIFI ?
Inv. with integrated WIFI are not supported

PaTiTan, 2025-06-21:
https://github.com/lumapu/ahoy/commit/fa8d16b51d652dfdf863223b979f7d63d74f879a

I tried to make the modification, it works correctly, I have all the information in the ‘live’ but I can't add the MAX of the different panels in the HTML.

thoniodu10, 2025-08-04:
Hi,

Same issue here, looking at the code, it seems that HMS-1000-2T is not supported.
Could be great to add it.

Thanks!

chaos3636, 2025-08-13:
Hi,

Unfortunately, version 0.8.156 is not yet functioning for me. Inverter is displayed, but no data is being received. Update was performed via web installer.

[image]

[image]

[image]

chaos3636, 2025-08-13:
[image]

[image]

Bei mir laufen aktuell 5 Wechselrichter, wovon der HMS-1000 Nummer 6 ist. Hier sind auch zwei HMS-2000 die ohne Probleme laufen. Entfernung sollte kein Problem sein.

chaos3636, 2025-08-13:
Wenn ich alle 3 auf 868mhz stelle finde ich nur einen HMS-2000 und einen HMS-2000 & HMS-1000 nicht. Habe jetzt bissl gespielt und eine Variante gefunden, die zu funktionieren scheint. Die beiden HMS-2000 sind wieder auf 863mhz (default) und der HMS-1000 auf 867mhz (868mhz hat nicht funktioniert).

Jetzt funktioniert es, danke für den Hinweis!

[image]

PaTiTan, 2026-11-09:
Follow-up after testing 0.8.156 with two HMS-1000-2T (`1410…` serials) on an ESP32-WROOM32: the inverters were added, but showed "not yet available" ~95% of the time although they were producing.

Found the cause: the new `0x14` branch in `hmSystem.h` doesn't set `ivRadioType` (stays UNKNOWN), so the CMT specific handling in `Communication.h` is skipped. On top of that `duration_reserve[2]` is indexed by a 3-value enum (out of bounds for CMT).

With both fixed the two inverters are reported as available and producing. Draft PR: https://github.com/lumapu/ahoy/pull/1908.