lumapu/ahoy #1849: inverter unavailable after minutes - manaul need to search for inverter

mkrasselt1, 2025-08-07:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

I have only a CMT module connected, mqtt to immediately and 3 Inverters with 6 Channels each.
2s Intervall and Homeassitant Discovery needs to bee resend every so often as well.

### Debug Serial Log output

```shell

```

### Error description

I used to stick to 0.8.83 for a while since it has better inverter stability (1s interval no problem for week)
But i could not re-add to home assistant because discovery was broken.
That why i updates to current version and got this problems. 
had this with the previous version as well and the web serial just says transmission error.
Under settings "Search for Inverter" brings them back.
But for now i will roll back to the previous version.
Setting a lower inverter poll rate does not help at all (tried 30s as extrem example)

kaindjoh, 2025-08-14:
same problem here. i have 2 hms-1600-4t, 1 inverter is not available after a few minutes with 0.8.156, the second one is ok. back to 0.8.154 helps. with 0.8.155 both inverters are not available.

datenschuft, 2025-08-20:
same problem here
Version 0.8.155 and version 0.8.156 lost connection to both of mine inverters (not at  the same time)
current system: esp32-wroom32-de (Module: MDH-de)
Version	0.8.156 - c4fe30f
Chip	CPU: 240MHz, 2 Core(s)
Chip Model	ESP32-D0WDQ6
with NRF24 module
will try a downgrade to 0.8.154

Gubi2023, 2025-08-20:
ich würde die 0.8.152 empfehlen. Ist zwar keine release, aber läuft bei mir seit 3 Monaten anstandslos..

mkrasselt1, 2026-03-26:
> I would like to test the 0.8.152 or 0.8.153 dev branch, but I need to get the source code because I'm with ESP32C3Zero and I need to remove the serial wait loop to boot without the USB cable...

its here
https://github.com/lumapu/ahoy/tree/78cbd7749d1f81a5b71b0d2ea36c70c1d23b0a96