lumapu/ahoy #1852: AhoiDTU liefert nach mehreren Monaten plötzlich keine Daten mehr.

DeSeDo54, 2025-09-02:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

[image]

### Version / Git SHA: 
Version: 0.6.0
Github Hash: bb5cb8b

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [x] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

DeSeDo54, 2025-09-02:
Hallo zusammen,
habe mich hete neu hier registriert und hoffe auf eure Expertise.
Vor mehr als zwei Jahren habe ich eine kleine PV-Anlage mit vier Modulen a 425 Wp und zwei Hoymiles HM600 installiert und eine AhoiDTU zusammengebaut. Über die AhoiDTU wurden dann die beiden HM600 ausgelesen und gesteuert.

Seit ein paar Wochen zeigt die AhoiDTU nur noch Null-Werte an. Dazu habe ich diverse Fotos gemacht, um vielleicht zu erkennen, wo etwas nicht passt. 

[image]
[image]

Die Konfiguration ist unverändert (hat ja auch mehr als zwei Jahre funktioniert) und auch die Bauteile sind seit dem Zusammenbau nicht ausgetauscht.

[image]
[image]
[image]
[image]
[image]

Würde mich über Hinweise und Tipps freuen, die mich in die richtige Richtung und zu einer funktionierenden AhoyDTU bringen. Besten Dank schonmal.

Beste Grüße,
DeSeDo54

DeSeDo54, 2025-09-02:
Hallo mroenne2022,
ja, fängt bei mir auch mit 1141... an. Habe ich vor veröffentlichung unkenntlich gemacht.

mroenne2022, 2025-09-02:
Mit dem Update hat das nichts zu tun glaube ich, wie ganz oben zu lesen hat er noch 0.6.0 auf der DTU,  der Update wird nur angezeigt das er verfügbar ist.

mroenne2022, 2025-09-03:
Hallo DeSeDo54 , warum du kein Signal von den WR bekommst ist mir noch immer nicht klar !?
Oben hab ich gelesen das die DTU die Version 0.6.0 hat ? Ist das noch so oder hast du mal ein Update gemacht ?
Warum ich frage .... Das Problem mit dem dauerhaften leuchten der blauen LED ist bekannt und trat auch bei mir und meinen
esp8266 wemos d1 mini auf - allerdings kam das mit dem Update auf Version 0.8.83 zum tragen.
Abhilfe hat seinerzeit bei mir folgendes geholfen : 
Einfach die Verbindungen D3 und D4 tauschen , das sind  die Ports CE und IRQ.
Bei mir sieht es seit dem wie folgt aus, seit dem war das Problem mit der blauen permanten LED erledigt,
aber wie gesagt - das kam bei mir erst mit v0.8.83 , aber dennoch kannst du das ja mal leicht testen !?

**CE/EN  auf GPIO00/D3*
IRQ auf  GPIO02/D4***

DeSeDo54, 2025-09-04:
Hallo mroenne2022,
aktiv habe ich kein Update gemacht und unten rechts in der Web-Anzeige steht als Version 0.6.0 und der o.g. Hash-Code bb5cb8b.

Eigentlich stört mich die blaue LED nicht, wenn die DTU ordentlich funktionieren würde. Soweit ich mich aber erinnere, hat eine (diese) LED geblinkt, vermutlich wenn Daten übertragen wurden.

Würde es evtl. was bringen, wenn ich unter System ein Factory Reset mache oder weden da nur die Konfigurationseinträge gelöscht?

mroenne2022, 2025-09-04:
Hallo , ein Factory Reset wird alle Konfig Daten löschen, aber du kannst die ja vorher sichern / exportieren - wenn die Version 0.6.0 das schon kann, bin ich nicht mehr sicher, und hinterher wieder importieren.
Aber ich empfehle dennoch den Tausch der beiden Ports zunächst mal vorzunehmen und zu testen ob es dann läuft, ist ja nicht so aufwendig.

DeSeDo54, 2025-09-06:
Hallo zusammen,
heute war Basteltag und um es vorweg zu sagen: "Das Problem sitzt vor der Tastatur" und das bin in diesem Falle ICH :-(

Dem Vorschlag von mroenne2022 folgend wollte ich heute die Ports tauschen und als Vorbereitung habe ich diverse Ausdrucke und handschriftliche Aufzeichnung herausgeholt, um zu identifizieren, wo die entsprechenden Ports sind.

Dabei ist mir aufgefallen, daß der Port D3 nicht mit IRQ verbunden ist, sondern bei mir der Port D1.
Nachdem ich dann in der Konfiguration dies entsprechend zugeordnet hatte, waren auch wieder Daten vorhanden.

Da das System seit Anfang/Mitte 2023 läuft, kann ich mich nicht mehr erinnern, warum das bei mir so ist.
Hauptsache aber: Es funktioniert wieder.

Nochmal besten Dank an alle, die sich zu meinem Problem Gedanken gemacht haben.

Beste Grüße,
DeSeDo54