lumapu/ahoy #1858: Can't enter serial number with letter "A"

siwakornp1-ship-it, 2025-09-25:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

board

### Connection picture

- [x] I will attach/upload an image of my wiring

### Version

 0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT: only broker was added
Inverter: set interval to 5 seconds
(Checked via: http://dtu.setup/setup)
No private data (SSID/password/serial) is included.

### Debug Serial Log output

```shell
[INFO] Starting AhoyDTU v0.88.3
[INFO] GitHub Hash: 5ebfe5a
[INFO] Using Arduino IDE for build
[ERROR] Invalid character in serial number input: 'A'
[WARN] Serial number rejected due to format
```

### Error description

The software does not allow me to enter a serial number that contains the letter "A".
For example, when I try to input 1164A......., the field either blocks the input or does not save it correctly.

[image]

stefan123t, 2025-09-25:
@lumapu this is a modern year code, we should accept these in Ahoy too.

**Micro-inverter system related product ID coding rules**
https://github.com/lumapu/ahoy/issues/1111#issuecomment-2444887377

> The 5th digit is the year of production (time provided by the welder), the `1` is for 2015, and so on

5th digit | Production Year
-- | --
1 | 2015
2 | 2016
3 | 2017
4 | 2018
5 | 2019
6 | 2020
7 | 2021
8 | 2022
9 | 2023
A | 2024
B | 2025
C | 2026
D | 2027
E | 2028
F | 2029

siwakornp1-ship-it, 2025-09-25:
> Hallo,
> 
> die Version 0.8.83 ist uralt und sollte aktualisiert werden. Dazu solltest du das Board komplett neu flashen, da es mehrere grundlegende Änderungen gab.
> 
> Nachdem die DTU aktualisiert wurde, funktioniert vermutlich auch die gewünschte Seriennummer. Sollte es dann wieder erwarten immer noch nicht funktionieren, benötigen wir die ersten 6 Stellen der Seriennummer.
> 
> LG

siwakornp1-ship-it, 2025-09-25:
Thank bro it work

closed 2025-09-25 by siwakornp1-ship-it